soligen2010/Adafruit_ADS1X15 #1: Conversion Delay Issues

soligen2010, 2016-02-10:
This is a continuation of an issue from adafruit's fork.  I created this issue to continue the discussion on addressing the issue because it is being worked on in fork. the original issue is [here](https://github.com/adafruit/Adafruit_ADS1X15/issues/14)

Summary:  Conversion delays have been removed in favor of polling the chip every 1 ms until conversion completes. The delay(1) causes a yield for ESP8266 to it doesn't eat up too many cycles.

Here is the final post from the old thread

It occurs to me that 1 ms delay resolution makes the faster SPS settings rather pointless.

gordonthree, what specifically is your concern of taking too many cycles? I am considering changing to delay(0) which will tighten the loop and make response faster, but will still do a yield for ESP8266 processing. Thoughts?

gordonthree, 2016-02-10:
A few steps forward, and maybe one backward? I tested up to 250SPS and had no problems with the module's output, all the numbers were reading for their respective ports and only those ports.

However I seem to have lost the gain setting, or bit resolution. Using a 10k pot, I used to get very small changes registering right away as I turn the pot from 3.3v down to 0v. Now, no change registers until the pot is in the 2v range, and then it drops quickly.
Using this in setup:
  ads.begin();
  ads.setGain(GAIN_ONE); 
  ads.setSPS(ADS1115_DR_128SPS) // also tried 32, 64, 250, same result

I'm switching back to the previous version to see if I messed something else up, or if its in the library.

gordonthree, 2016-02-10:
ignore previous post, I had stupidly wiring the pot wrong. 

this is working great at 250SPS. I don't really know how to test it even at that speed, that's way faster than the serial monitor.

My application is reading the analog output of an op-amp that's monitoring a soil moisture probe, so even the slowest setting on the ADC is more than enough for me.

soligen2010, 2016-02-11:
Glad its working.  I reduced the delay in the wait loop (still yields each iteration).  Latest version is uploaded. I also have put the sketch I use for testing in the examples.

soligen2010, 2016-02-11:
I am closing this issue as I think it is resolved.  Feel free to open another issue if more attention is needed.

closed 2016-02-11 by soligen2010