soligen2010/Adafruit_ADS1X15 #3: Random spikes when running for a long time?

xiaoken, 2016-02-27:
Hi, I'm having a issue where the adc sent value seems to be not read sometimes... 
the adc is used to measure a voltage divider circuit, where a thermistor and a reference resistor are connected in series and a temperature is measured.
The measured resistance is first put into a array for getting the rolling average (reduces the noise), and sometimes the average of this array drops as shown below....
I used a fletcher16 checksum to make sure there were no errors...
now i'm just guessing there are times when the ADC is returning weird values....
Would be great if you could give me some advice.....thanks :)

[image]

soligen2010, 2016-03-01:
Ken,

Could be lots of things.  Seeing as it starts after a long time, it suggests something environmental - perhaps components warm up???? Maybe rig a fan to blow on the board to rule this out.  

If this is on a breadboard, connections may be dodgy. You can try soldering it up. 

Have you tried a different ADS device to see if maybe it is the chip itself?  We have noted on a different thread that the smaller purple boards take longer to do a conversion than the blue boards.  Could be that the cheap ebay modules have more issues than just this.

It is also possible your circuit itself is doing this?.  If you have an Oscilloscope, you can check and see if the chip is reflecting reality or not.  A multi meter will likely be too slow to show any problem.  I also suggest looking at raw data rather than summed 500 times.

In your code I see you are changing the readings to unsigned. Even in single ended mode you can get negative numbers, which may cause you grief.  Whenever I try to read ground, it usually shows up as slightly negative. Even if its not the problem, if you never expect or want negative voltages I would suggest you add right after the read:

if (voltage01 < 0) {voltage01 = 0};

BTW, waitForConversion is built into the single ended and differential reads, so you don't need to call it.  It was made public in case it was needed when using the comparator mode. 

I don't know what is wrong, but I hope some of these thoughts help.  Please post back your progress.  I am interested to see how this turns out.

xiaoken, 2016-03-02:
Ya, this is laid out on a breadboard, but i have some trouble reproducing the situation,
the data shown in the previous post was taken once per 100ms.
In the past 3 days, i tried getting the measured voltage along with the averaged value "every loop",
but i haven't been able to obtain similar data.
the loop time will be almost tripled to around 20 ms if i send values every loop,
and will be around 8 ms if i send once 100 ms.

About the conversion time, since this is not a board made from adafruit, i suppose the conversion time could be a problem... i'll try changing the conversion time in the library if i run into this again.

My circuit is really simple so i don't suppose peaks like this could happen. 

About the voltage01 part, good thing i don't measure values near 0V :)
I'll keep that in mind though :+1:  thanks!

soligen2010, 2016-03-02:
My latest version doesn't use conversion delay.  I now use the registers on the chip to determine when conversion is complete instead of a fixed wait time, thus insulating the library from slow chips.  This basically lets the conversion time be as fast as possible based on the SPS and whatever other time the chip needs.  It doesn't seem like this would cause your symptoms, but if you don't have the latest version, might want to try it just in case.

soligen2010, 2016-03-09:
Ken,

Did you resolve the issue?  I am curious about this.  Also, if it is not software related, I should close this issue.

xiaoken, 2016-03-09:
Yah no more spikes... for no specific reason, thanks for your help :)

closed 2016-03-09 by xiaoken