soligen2010/Adafruit_ADS1X15 #6: Can not access I2C addr 0x4A and 0x4B with ADS1115 and Cortex M0

HamishB, 2017-07-12:
Hi,

I've got a number of ADS1115 boards which I've been trying to get to work with Adafruit Feather Cortex M0 Adalogger Arduinos. With the stock Adafruit ADS1X15 library none of the single ended outputs lined up, Differential mode 01 showed up on 23 and other craziness -- with your fork it Just Works. Brilliant, thanks!

Using an ATMega328p the original library from Adafruit worked fine, just not with the faster M0.

One problem I'm still having though, that was also seen with the original library is that I can only change the I2C address between 0x48 and 0x49 (addr tied to gnd or vcc) but I get no response if I try addresses 0x4A and 0x4B (addr pin tied to sda or scl). I've tried this with a number of ADS1115 boards and it's the same with all of them. I assume it's some timing conflict with the I2C lines? I have not tried adding a pF capacitor there, but it's an idea.

Have you/anyone else had a similar setup working on the two higher I2C addresses? This issue limits me to using two ADS boards at a time instead of four.

many thanks for your fine work,
Hamish

soligen2010, 2017-07-12:
I have not tries the higher addresses, but here are some things that come to mind.  Each module has it;s own pull up resistors.  If using multiple modules you may need to remove them from all but one module or the pull up may be too strong.

Make sure you are trying the higher addresses with just one module connected to be sure it is not other modules interfering.

When wired for a higher address, try the other addresses to see if it is responding, just on the lower addresses.

I am not sure about adding a capacitor to SDA or SCL.  That may mess up the i2c timing, but experimentation is part of the fun :)

HamishB, 2017-07-12:
> Make sure you are trying the higher addresses with just one module connected to be sure it is not other modules interfering.

> When wired for a higher address, try the other addresses to see if it is responding, just on a different addresses.

Yes, I'm trying with only one module at a time, and everything works fine with the lower two I2C addresses on the same module.

I'll experiment some more today.

thanks

HamishB, 2017-07-13:
Ok,

with two ADS1115 boards on the I2C bus I can successfully set either of the boards to use the high 0x4A and 0x4B addresses, but with only one board attached on the bus I am limited to the lower 0x48 and 0x49 addresses only.

with three ADS1115 boards on the I2C bus all four addresses work as well. I didn't have enough room on my breadboard to test with four ADS1115 boards.

So the problem seems to be accessing the high memory addresses when there is only one board connected. I guess that's usually not needed, and if it were needed changing out the pullup resistors or putting additional resistors in parallel could be the solution?

Anyway, probably not a bug in the library, so closing.

thanks.

closed 2017-07-13 by HamishB