soligen2010/Adafruit_ADS1X15 #7: no conversion

chseiler, 2017-11-07:
i am having troubles using the adc1115 with your library. no matter what i do, i always get a reading of -1 for all channels. can you help me?

soligen2010, 2017-11-07:
Please try the single ended example and see if that works OK, and post the details of what you are doing.

soligen2010, 2017-11-07:
Also please be sure to read all of adafruit's instructuions at
[https://learn.adafruit.com/adafruit-4-channel-adc-breakouts](https://learn.adafruit.com/adafruit-4-channel-adc-breakouts)
Make sure your wiring is correct. Be sure to tie the ADDR pin to ground to set the I2C address

chseiler, 2017-11-07:
Thanks for your fast replies:

The problem was, that my I2C port on the arduino was broken and I realized it just after I wrote the message. Your library works like a charm on a new arduino due.

All the best
Christian

https://learn.adafruit.com/adafruit-4-channel-adc-breakouts<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Flearn.adafruit.com%2Fadafruit-4-channel-adc-breakouts&data=02%7C01%7Cchristian.seiler%40helbling.ch%7C6883211ea41041fe46a508d525ee46c4%7Cbcbab93b0f0d4747b4a4bc57e6753bb4%7C0%7C0%7C636456627677498523&sdata=tdAkUfPwi3qAheppA5fpcvPyRQCq6DZiu7FR1gQOT2w%3D&reserved=0>

closed 2017-11-07 by soligen2010