soligen2010/Adafruit_ADS1X15 #8: timeout if no connection to address available

eimix, 2018-01-15:
I use your modified library - it makes adc read time as short as possible.

But sometimes i get my ESP8266 stop responding. And problem disapears if i do not use ADC.
Could it be that ADC reading loop (waitForConversion) blocks everything, if connection to ADS1105 is lost, or ADS1105 is not responding?

soligen2010, 2018-01-16:
I have not really used the ESP with the ADC other than the example program.  Wait for conversion has delay(0) in there for the ESP8266 because delay does a yield.  Maybe the zero duration prevents the yield.  You can try replacing the delay(0) with yield instead and see if that helps.

As far as why the adc seems to stop responding, nothing software related comes to mind.

soligen2010, 2018-01-16:
Another option is if you are only sampling one input (or one differential) with the ADC, then you can use continuous mode. After the first priming read there is no wait loop. You just keep asking for the value and it gives the result from the most recent conversion.

If you decide to go this route, I would appreciate it if you can still test the change to using yield and see if that helps so the library can be updated if it works.

eimix, 2018-01-16:
delay(0) is ok, it has yeild, so ESP8266 keeps responding to ping, but program is not executed any more.

This `do {} while ` looks like infinite loop if connection to ADS11X5 is lost. 

```
void Adafruit_ADS1015::waitForConversion()
{
  delay(0);                // delay(0) causes a yeild for ESP8266
  delayMicroseconds(10);   // Slight delay to ensure converstion started.  Probably not needed, but for safety
  do {
	  delay(0);            // delay(0) causes a yeild for ESP8266
	 } 
	 while (ADS1X15_REG_CONFIG_OS_BUSY == (readRegister(m_i2cAddress, ADS1X15_REG_POINTER_CONFIG) & ADS1X15_REG_CONFIG_OS_MASK));
            // Stop when the config register OS bit changes to 1
} 
```
Should it have some kind of time out?

In my case it is 3 phase current transformers (power monitor), i doubt that continuous mode is possible.

soligen2010, 2018-01-17:
Yes it is if the conversion never happens, but the conversion should always happen. Seems like a hardware issue is likely.  You could add a time out to waitForConversion(), but this wont fix the real underlying issue.  I did one project with the ESP and this ADC and never had this issue.  From the software side, the communication with the ADC uses the I2C library, so a bug in that library is possible too (but I think this is unlikely)

If it is an intermittent I2C communication issue, perhaps the request for the conversion is getting lost. and the chip never got the command

Not knowing how experienced you are with the hardware side, here are some things to look at.

Make sure your wiring is solid and short.  Breadboards can easily have bad connections.
Make sure your power supply is solid and can provide enough current.
Try a different ADC.  The market is flooded with cheap clone chips of dubious quality.  I use the clones and have had bad chips.
If you are using multiple I2C modules and each has it's own pull up resistors, it could be too much pull up and some hardware mods are needed.

soligen2010, 2018-07-09:
No comments in a while so closing the issue

closed 2018-07-09 by soligen2010