soligen2010/Adafruit_ADS1X15 #11: Question regarding differential

hicotton02, 2018-10-02:
First off, thank you for making this fork. I am implementing it into my project and have a couple questions.

I have 2 hall effect sensors plugged into 0 and 1, and 2, and 3. they both go from 0-2V. I am trying to use an ads1015 as an ammeter (connected to an esp8266)

I would like it to continually convert, and allow me to sample the conversion as the esp has time. the esp is doing a lot of other stuff. So i was thinking doing continuous conversion + differential on 0_1 and 2_3.
I dont really need the alert pin, just need it to be ready when the data is asked for.

first, does that make sense? 
if it does, ill post the code I have now (using your library fork) to try and get the values.

soligen2010, 2018-10-03:
You will need 2 chips running on 2 different I2C addresses to do this. You
cant run 2 different continuous conversions at the same time.

soligen2010, 2019-01-21:
Closing - no activity

closed 2019-01-21 by soligen2010