soligen2010/Adafruit_ADS1X15 #12: i2C high-speed mode master code

datablacksmith, 2019-01-18:
Hi soligen-

Love this library, thanks for writing it.

I'm trying to push the limits of the ADS1015 sampling speed. I'd like to put the ADS1015 into I2C high-speed mode (3.4MHz). This thread http://e2e.ti.com/support/legacy_forums/automotive/f/26/t/552044 suggests that upgrading from I2C fast mode (400 kHz) to high speed mode (3.4MHz) may save up to 77us on the read time, which would be significant in my application. Even half or a third of that would be awesome.

From the datasheet http://www.ti.com/lit/ds/symlink/ads1015.pdf :
> No special action is required to use the ADS101x in standard or fast mode, but high-speed mode must be activated. To activate high-speed mode, send a special address byte of 00001xxx following the START condition, where xxx are bits unique to the Hs-capable master. This byte is called the Hs master code, and is different from normal address bytes; the eighth bit does not indicate read/write status. The ADS101x do not acknowledge this byte; the I2C specification prohibits acknowledgment of the Hs master code. Upon receiving a master code, the ADS101x switch on Hs mode filters, and communicate at up to 3.4 MHz. The ADS101x switch out of Hs mode with the next STOP condition.

Any idea how to implement this? I looked through your library, but I'm flummoxed because:
- I cant figure out how to send a binary byte like "0000 1111" over I2C
- Does the Hs master code get written to a register, if so, which one?

I'm using a microcontroller that is high speed mode capable, the ATSAMD51.

Seems like this would probably end up being 3 lines of code, but I'm an Arduino noob, sorry.

Thanks for any help you can offer!

soligen2010, 2019-01-21:
I didn't write the I2C stuff, but there is an i2cwrite method you can try. You will need to change it to be public.

To get best speeds you can try continuous mode - it is the only way you will get the fully specified samples per second.  You can either use an interrupt to tell when each conversion is done, or just read the value whenever you like and you will always get the most recent conversion(note you could get the same reading twice if there is not a new conversion since the last read).  With continuous mode you save all communication overhead spent polling the chip to see if the conversion is done.  Downside is that in continuous mode you can only read one pin (or differential) and will need multiple chips if you need more than one input.

Also, if speed is a concern, I have encountered some Chinese counterfeit chips that run slower than the spec.  If this happens to you, you may want to try legit chips.

Cool user name.  I am a Data Architect.  Are we in the same field?

soligen2010, 2021-02-10:
Closing due to inactivity.

closed 2021-02-10 by soligen2010