soligen2010/Adafruit_ADS1X15 #13: Detect that ADS1x15 is not present?

Humancell, 2019-01-23:
Hello,

Thank you for taking this library forward.  I appreciate the work you have put into it ... Adafruit owes you for doing their work for them.  :-)

I noticed that if the ADS1x15 is NOT present on the I2C bus, there is no indication that I can find that my code can use to know the values are all "bad".  Often, in other libraries, the .begin() method will fail if the chip is not detected on the I2C bus.

Is there currently a way in the library to determine that an ADS1x15 chip is or is not present and responding?

Do you have ideas or suggestions on implementing such a test?

Thanks again!

soligen2010, 2019-01-23:
I don't think this is implemented.  I guess this could be done if you can find a command to send the chip that has a predictable response.

I have renewed hopes that Adafruit will merge this into their base, in which case you could make this addition to the original library.  I don't think I would take a pull request right now in hopes that they are going to merge what I already have.  However, you could add it to your own fork then send it in once we see where things land.

Humancell, 2019-01-23:
Yes ... I looked through the code and hadn't spotted anything.  There are a few easy ways this can be added:

1. When instantiating the class, a simple Wire test can be done:
```
Wire.beginTransmission(i2cAddress);
error = Wire.endTransmission();

```
The return value "error" will be 0 if there is a device at that address.  Not necessarily an ADS1x15, but at least a device is present.

2. I also noticed that the readRegister() and writeRegister() are not checking the return values for Wire.endTransmission() which might be returning errors that are simply being ignored.

I'm early in my testing, but might fork if I decide to implement one or both of these "fixes".

I did reach out to Adafruit to ask them about these libraries last week.  That is when I found your fork.  They explained about the overhead of maintaining libraries, and having to do regression testing, and testing across multiple boards.  They asked if I'd like to "take care" of a repo.  I'm thinking about what I want to ask for in return.  :-)

trycoon, 2019-08-17:
Any response from Adafruit?

trycoon, 2019-08-21:
Well they are not famous for maintaining their libraries... 😕
Maybe you could maintain this fork?